Lucaslhm/Flipper-IRDB #11: Documentation of compatible devices?

Patronics, 2022-05-02:
Is there any documentation of devices known to work with a given signal? That could be useful to list in a Readme for each directory. For example, I discovered that the "Zenith AC" turns out to be identical to the signals from my Kenmore 580-series Air conditioning unit.

Lucaslhm, 2022-05-02:
At the moment no, the community has just been adding remotes they scan with their flippers so we don't have much of a formal intake process or documentation process atm. In the future, I'd be interested in doing this, but I'm not entirely sure how plausible it would be in our current implementation.

closed 2022-05-02 by Lucaslhm

sundhaug92, 2022-09-21:
Was thinking about this recently. My idea is either:

1. Files are named by the remote, rather than the device it controls, or
2. Make the remote a metadata-field