Lucaslhm/Flipper-IRDB #12: Different select code for LG TV

makeworld-the-better-one, 2022-05-04:
Currently the code the Select button for LG C1 is this: https://github.com/Lucaslhm/Flipper-IRDB/blob/308cd339da9d2c0195ad212bfa15ca7f75328607/TVs/LG_C1.ir#L58-L62

All the buttons in that file worked with my LG TV, except the Select button. I recorded my Select button and got this code:

```
name: Select
type: parsed
protocol: NEC
address: 04 00 00 00
command: 44 00 00 00
```

I don't have an LG C1, but all the other buttons worked... maybe the LG C1 would respond to my Select button as well? If someone could test that'd be great, and then I could make a PR swapping mine in. Otherwise I'll make a separate file for my model, the LG 42LF5800 (Smart LED TV).

UberGuidoZ, 2022-06-06:
Added downstream:
 - https://github.com/logickworkshop/Flipper-IRDB/compare/main...UberGuidoZ:main

Now part of pending pull request:
 - https://github.com/Lucaslhm/Flipper-IRDB/pull/31

UberGuidoZ, 2022-07-26:
Merged as part of #31 above. Thank you! Closing.

closed 2022-07-26 by UberGuidoZ

jjfeldcher, 2022-12-21:
The file was not updated, but the code is correct.