Lucaslhm/Flipper-IRDB #48: invalid filenames

GottZ, 2022-07-27:
there are several files in _Pronto_Converted_ that contain `,` in their names.

flipper doesn't really like that character in file names.

UberGuidoZ, 2022-07-27:
We've been going through such a massive renaming but have not gotten to all of Pronto and CSV. Please, don't be shy about a massive rename if you'd like to help. =) Otherwise, we'll get to them as time allows. Leaving this open for now. Thank you!

zigad, 2022-07-28:
@UberGuidoZ @GottZ I have removed all of commas from file names and replaced them with "_" or where "_" was already present just removed them. It's quite a massive rename but I think this will be helpful in future renaming of single files.

UberGuidoZ, 2022-07-28:
Verified and merged. Thank you!

closed 2022-07-28 by UberGuidoZ