Lucaslhm/Flipper-IRDB #57: Index

nepnym, 2022-07-28:
Given the sizes of the pronto and csv databases there should be a file index of some sort with brands and serial numbers. This could probably be auto-generated, but the amount of research that would need to be done to correctly label a large part of the items would be massive. I would totally do this all just unprompted, but I know there are other IR database projects on github and in general that may have already done some of the legwork.

UberGuidoZ, 2022-07-28:
Great thought. For the Flipper specifically, the only 3 active ones I am aware of are this one (OG), my own (Rebirth), and Logick (created so mine could feed into one maintained by others.) Since I currently maintain all 3, I am not aware of this index existing. BUT I would love to have it. =) Please feel free to take a peek and see what you can do. For now, I simply search either via git or a download to see if something exists. ¯\_(ツ)_/¯

UberGuidoZ, 2022-07-31:
One thing that helps this a bit is a reorg that was just completed. Now you can browse by number or letter instead of the entire dump at once. Bit easier on the Flipper screen. =)

closed 2022-08-20 by UberGuidoZ