Lucaslhm/Flipper-IRDB #198: SR-7000.ir RC-5 parsed

wdoekes, 2022-09-22:
Hi!

Here are the regular RC-5 commands from https://github.com/Lucaslhm/Flipper-IRDB/blob/28ae99df0f695a5dcfa22b237be0f6802b2da9bd/_Converted_/IR_Plus/M/MARANTZ/SR-7000.ir

I skipped the RC-5 Ext codes, as I did not implement a parser for that yet. (All the raw ones which end in 7xxxx.)

Thanks for this repo. I used VOL+/VOL- to figure out how to this translates to the known (to me) 0x10/0x11 codes.

By the way, Marantz has a command sheet here:
https://www.marantz.com/-/media/files/documentmaster/marantzna/us/marantz-2014-ir-command-sheet.xls

So, that second Unknown here with 0x50 (80) is "CURSOR Up"

```
Filetype: IR signals file
Version: 1
#
# Unknown: 0b11110000001100
name: Unknown
type: parsed
protocol: RC5
address: 10 00 00 00
command: 0C 00 00 00
#
# VOL+: 0b11010000010000
name: VOL+
type: parsed
protocol: RC5
address: 10 00 00 00
command: 10 00 00 00
#
# Unknown: 0b10010000010000
name: Unknown
type: parsed
protocol: RC5
address: 10 00 00 00
command: 50 00 00 00
#
# CH+: 0b11000000100000
name: CH+
type: parsed
protocol: RC5
address: 00 00 00 00
command: 20 00 00 00
#
# Unknown: 0b10010000010101
name: Unknown
type: parsed
protocol: RC5
address: 10 00 00 00
command: 55 00 00 00
#
# OK: 0b10010000010010
name: OK
type: parsed
protocol: RC5
address: 10 00 00 00
command: 52 00 00 00
#
# Unknown: 0b10110000010110
name: Unknown
type: parsed
protocol: RC5
address: 10 00 00 00
command: 56 00 00 00
#
# VOL-: 0b11110000010001
name: VOL-
type: parsed
protocol: RC5
address: 10 00 00 00
command: 11 00 00 00
#
# Unknown: 0b10110000010001
name: Unknown
type: parsed
protocol: RC5
address: 10 00 00 00
command: 51 00 00 00
#
# CH-: 0b11100000100001
name: CH-
type: parsed
protocol: RC5
address: 00 00 00 00
command: 21 00 00 00
#
# CLEAR: 0b11110001111010
name: CLEAR
type: parsed
protocol: RC5
address: 11 00 00 00
command: 3A 00 00 00
#
# MEMO: 0b11010001101001
name: MEMO
type: parsed
protocol: RC5
address: 11 00 00 00
command: 29 00 00 00
#
# Unknown: 0b11110000001101
name: Unknown
type: parsed
protocol: RC5
address: 10 00 00 00
command: 0D 00 00 00
#
# OSD/9: 0b11010000001111
name: OSD/9
type: parsed
protocol: RC5
address: 10 00 00 00
command: 0F 00 00 00
#
# SLEEP/: 0b11110000100110
name: SLEEP/
type: parsed
protocol: RC5
address: 10 00 00 00
command: 26 00 00 00
#
# MODE: 0b11110000100101
name: MODE
type: parsed
protocol: RC5
address: 10 00 00 00
command: 25 00 00 00
#
# TREBLE+: 0b11010000011000
name: TREBLE+
type: parsed
protocol: RC5
address: 10 00 00 00
command: 18 00 00 00
#
# TREBLE-: 0b11110000011001
name: TREBLE-
type: parsed
protocol: RC5
address: 10 00 00 00
command: 19 00 00 00
#
# BASS+: 0b11010000010110
name: BASS+
type: parsed
protocol: RC5
address: 10 00 00 00
command: 16 00 00 00
#
# BASS-: 0b11110000010111
name: BASS-
type: parsed
protocol: RC5
address: 10 00 00 00
command: 17 00 00 00
```

wdoekes, 2022-09-22:
And here are the others.

Note that `protocol: RC5Ext` is not a thing (yet) in the Flipper.
```
#
# POWER: 0b11010000001100000001
name: POWER
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 0C 01 00 00
#
# OFF: 0b11110000001100000010
name: OFF
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 0C 02 00 00
#
# AUTO/1: 0b11010000100101101101
name: AUTO/1
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 2D 00 00
#
# DOLBY/2: 0b11110000100101000000
name: DOLBY/2
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 00 00 00
#
# THX/3: 0b11010000100101100100
name: THX/3
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 24 00 00
#
# DTS/4: 0b11110000100101101110
name: DTS/4
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 2E 00 00
#
# HALL/5: 0b11010000100101000001
name: HALL/5
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 01 00 00
#
# MATRIX/6: 0b11110000100101001001
name: MATRIX/6
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 09 00 00
#
# MOVIE/7: 0b11010000100101011101
name: MOVIE/7
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 1D 00 00
#
# 2CH/8: 0b11110000100101011110
name: 2CH/8
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 1E 00 00
#
# A/D: 0b11010000000001000001
name: A/D
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 01 01 00 00
#
# ATT: 0b11110000100101101011
name: ATT
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 2B 00 00
#
# NIGHT: 0b11010000100101101010
name: NIGHT
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 25 2A 00 00
#
# MULTI: 0b11010000011101000000
name: MULTI
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 1D 00 00 00
#
# 6-DIR: 0b11110000000001000011
name: 6-DIR
type: parsed
protocol: RC5Ext
address: 10 00 00 00
command: 01 03 00 00
```

wdoekes, 2022-09-22:
(Note that that is not RC5X (in the flipper), which apparently has 7 bit commands, but a Marantz-specific extension.)

UberGuidoZ, 2022-09-23:
Thank you for the info! I know some of the converted files we have created are not compatible (yet?) with the Flipper firmware. The hope is it will eventually be implemented! If you have additional buttons for existing files we'd love to see a PR. =)

wdoekes, 2022-09-24:
Personally, I was looking for these only:
```
#
name: POWER ON/OFF
type: parsed
protocol: RC5
address: 10 00 00 00
command: 0C 00 00 00
# INPUT NEXT [16 0 13] {110-10000--000000-001101}
name: INPUT NEXT (raw)
type: raw
frequency: 36000
duty_cycle: 0.25
data: 889 889 1778 1778 1778 889 889 889 889 889 889 4445 889 889 889 889 889 889 889 889 889 889 889 889 889 889 889 1778 889 889 1778 1778 889 75551
# INPUT BACK [16 0 14] {110-10000--000000-001110}
name: INPUT BACK (raw)
type: raw
frequency: 36000
duty_cycle: 0.25
data: 889 889 1778 1778 1778 889 889 889 889 889 889 4445 889 889 889 889 889 889 889 889 889 889 889 889 889 889 889 1778 889 889 889 889 1778 76440
#
name: TV AUDIO(TV)
type: parsed
protocol: RC5
address: 00 00 00 00
command: 3F 00 00 00
#
name: CBL/SAT(SAT,VCR2)
type: parsed
protocol: RC5
address: 06 00 00 00
command: 3F 00 00 00
```
So, now I'm content :)

closed 2022-09-24 by wdoekes