Lucaslhm/Flipper-IRDB #261: Sony Bravia 4K VH21 KD-55X80CK

yasbean, 2022-11-24:
I could not figure out how to contribute my control so I am putting it here:

```
Filetype: IR signals file
Version: 1
# 
# Sony KD-55X80CK Bravia 4K VH21 TV
#
name: POWER
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 15 00 00 00
# 
name: MUTE
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 14 00 00 00
# 
name: VOL+
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 12 00 00 00
# 
name: CH+
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 10 00 00 00
# 
name: VOL-
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 13 00 00 00
# 
name: CH-
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 11 00 00 00

```

yasbean, 2022-11-24:
And here are some more additions not included in the standard format:

```
name: Input
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 25 00 00 00
# 
name: Enter
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 65 00 00 00
# 
name: Left
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 34 00 00 00
# 
name: Right
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 33 00 00 00
# 
name: Up
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 74 00 00 00
# 
name: Down
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 75 00 00 00

```

Schecher1, 2022-11-24:
Hey, should i create a pull request for you?

yasbean, 2022-11-24:
Please. I am not very GitHub literate and know only how to grab stuff and open issues.

Schecher1, 2022-11-24:
All right, I'll create one with your information. Thanks!

yasbean, 2022-11-24:
Re-labeled the additional buttons to a more readable format.  (Sorry for the inefficiency!)

```
Filetype: IR signals file
Version: 1
# 
# Sony KD-55X80CK Bravia 4K VH21 TV
#
name: POWER
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 15 00 00 00
# 
name: MUTE
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 14 00 00 00
# 
name: VOL+
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 12 00 00 00
# 
name: CH+
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 10 00 00 00
# 
name: VOL-
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 13 00 00 00
# 
name: CH-
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 11 00 00 00
#
name: INPUT
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 25 00 00 00
# 
name: OK
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 65 00 00 00
# 
name: <- LEFT
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 34 00 00 00
# 
name: RIGHT ->
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 33 00 00 00
# 
name: UP  ^
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 74 00 00 00
# 
name: DOWN  V
type: parsed
protocol: SIRC
address: 01 00 00 00
command: 75 00 00 00

```

Schecher1, 2022-11-24:
Thanks, but I think the "direction names" were enough, is also everywhere else so. Personally, I would even say that the arrows would be annoying.

yasbean, 2022-11-24:
I was thinking of doing all arrows or just words. It actually looks better on the Flipper, but, hey, it's easy enough for folks to mod and personalize as they please.  Thanks for the pull request!

Schecher1, 2022-11-24:
It's "your" PR, if you like, I'll do it but I thought I think for all here, so that it remains nice and clean. Ah anyways no problem. 
Btw. My PR has to be checked by "UberGuidoZ" anyway. I have no rights, am actually a random who wanted to help you. :D

yasbean, 2022-11-24:
The way I see it, once posted, it is no longer mine.  Thank you, again, for your help!

closed 2022-11-24 by yasbean

UberGuidoZ, 2022-11-25:
Thank you both. Will get the PR merged.