Lucaslhm/Flipper-IRDB #278: Naming for AC buttons

airs0urce, 2022-12-12:
Hello. In README.md I see naming for ACs is:

> FOR AC
> 
> POWER
> TEMP+
> TEMP-
> MODE
> SWING

I have Electrolux ESV09CRO-B2I air conditioner.
There are couple issues:
1) Power buttons sends different signals for turning on and off. So, it must be two buttons POWER_ON and POWER_OFF
2) For modes there are 3 different buttons. SO, it will be MODE_FAN, MODE_AC, MODE_DRY

Is that ok if I'll use this naming?

Schecher1, 2022-12-12:
Thanks for your question/issue, 
it's kind that you ask if you can do that, actually if you look at some remote controls then you can see that the naming becomes inconsistent. 
In short, you are **allowed** to create other buttons, as long as they are **legible** and **understandable**. 
Thank you.

airs0urce, 2022-12-12:
I thought if later this database used for creating universal remote, buttons should be named the way possible to take correct signals for each button with script. Ok, Ill add couble more ACs soon :)

closed 2022-12-12 by airs0urce

Schecher1, 2022-12-12:
Don't forget, we're here on GitHub. Every second person can program here :D And as long as you name the buttons logically, you can do anything with programs.