Lucaslhm/Flipper-IRDB #285: Apple TV Gen 3 remote incorrect?

Wingman4l7, 2022-12-15:
This is what's been uploaded: https://github.com/Lucaslhm/Flipper-IRDB/blob/main/Streaming_Devices/Apple/Apple_TV_Gen3.ir
This is what I just made myself with one of these remotes:

https://dev.flpr.app/s#path=infrared/Apple_TV.ir&Filetype=IR+signals+file&Version=1&name=Menu&type=parsed&protocol=NECext&address=EE+87+00+00&command=02+2E+00+00&name=Play_Pause&type=parsed&protocol=NECext&address=EE+87+00+00&command=5E+2E+00+00&name=Up&type=parsed&protocol=NECext&address=EE+87+00+00&command=0B+2E+00+00&name=Down&type=parsed&protocol=NECext&address=EE+87+00+00&command=0D+2E+00+00&name=Left&type=parsed&protocol=NECext&address=EE+87+00+00&command=08+2E+00+00&name=Right&type=parsed&protocol=NECext&address=EE+87+00+00&command=07+2E+00+00&name=Center&type=parsed&protocol=NECext&address=EE+87+00+00&command=5D+2E+00+00

There appear to be some discrepancies.
I actually don't have one of these Apple TVs set up; anyone want to verify these?

UberGuidoZ, 2022-12-17:
I have no issue adding both so people can use what works. Apple is notorious for having slight differences between the same devices due to a firmware or version change.

UberGuidoZ, 2022-12-17:
Done! https://github.com/Lucaslhm/Flipper-IRDB/commit/c5e600aef0155606ddfc1a1f13ca3eb157f7cf69

closed 2022-12-17 by UberGuidoZ