Lucaslhm/Flipper-IRDB #323: Question

ninko76, 2022-12-27:
Would there be a way to, for example make all Sony tv's a universal remote option?

UberGuidoZ, 2022-12-28:
There certainly could - but it would take firmware modifications to create a Sony-specific Universal IR file that combines all Sony IR files into one, then has a "scene" for it (the remote you see). I'd recommend looking at the existing code of Universal remotes, including the tv.ir file, to see if it's something you want to take on. =)

mmiller7, 2022-12-31:
> There certainly could - but it would take firmware modifications to create a Sony-specific Universal IR file that combines all Sony IR files into one, then has a "scene" for it (the remote you see). [...]

I'm guessing that would be something to put in a feature-request for on the Flipper firmware page?

Would be neat if you could have it select a folder and it generates from there...or have some more config to the "universal remote" app.  I'm presently playing with a hotel TV which works on the "universal" remote but I'm yet to figure out what code actually is the right one.  It's a Sony Vizio...so being able to select by "all for X brand" would be a cool addition.

And similarly if it could aggregate the buttons - I want stuff like "Input/HDMI" that isn't on the hotel's remote in the room.  I know it's typically possible because I have a big clunky universal remote and booklet of directions/codes to program it for that was my go-to after smartphones eliminated the IR blasters.

I wonder - an alternative, could some script be run that combines all the sub-files commands into a string of commands that the Flipper would see and play as a single normal remote?

UberGuidoZ, 2023-01-01:
There are a handful of issues with this, at least, with current options. The universal button layout you see needs to be compiled into the firmware at the time of creating the firmware, so creating a custom one "on the fly" isn't possible. This is because of the layout and the manifest files required to make it work. Also, the universal remotes need to be culled through trial and error at times, and though are pulled from the IRDB (at least on some custom firmware), not every button works and some break the functionality entirely. It's a complicated process that, currently, involved 2-3 people manually fixing it and a handful of scripts that help check/automate it. But none of this runs from the Flipper directly.

We've looked into combining commands, say for "macros" where you can have a single command that turns on the TV, changes to a certain input, then sets the volume to a certain level. It has yet to work due to timing issues. It's a heavily requested feature something that would be liked, if possible. =)

closed 2023-01-03 by UberGuidoZ