Lucaslhm/Flipper-IRDB #410: Creating custom IR

Sigri44, 2023-02-16:
Hello,

I have pick an irplus file for my projector :
```irplus
<irplus>
  <device manufacturer="EXCELVAN" model="YG300" columns="12" format="WINLIRC_NEC1">
    <button label="&#1045541;" labelSize="25.0" span="4" backgroundColor="FFFF0000">
    	0xC03F 0xB847
    </button>
    <button label="" labelSize="12.0" span="4" fillColor="FF000000"></button>
    <button label="&#1045889;" labelSize="25.0" span="4">
    	0xC03F 0x40BF
    </button>
    <button label="&#1045599;" labelSize="25.0" span="4">
    	0xC03F 0xE01F
    </button>
    <button label="&#1045518;" labelSize="25.0" span="4">
    	0xC03F 0x9867
    </button>
    <button label="&#1045009;" labelSize="25.0" span="4">
    	0xC03F 0xA05F
    </button>
    <button label="" labelSize="12.0" span="4" fillColor="FF000000"></button>
    <button label="&#1044803;" labelSize="25.0" span="4">
    	0xC03F 0x906F
    </button>
    <button label="" labelSize="12.0" span="4" labelColor="FF000000" backgroundColor="FF000005"></button>
    <button label="&#1044801;" labelSize="25.0" span="4">
    	0xC03F 0x50AF
    </button>
    <button label="OK" labelSize="25.0" span="4">0xC03F 0xD02F</button>
    <button label="&#1044802;" labelSize="25.0" span="4">
    	0xC03F 0x10EF
    </button>
    <button label="" labelSize="12.0" span="4" fillColor="FF000000"></button>
    <button label="&#1044800;" labelSize="25.0" span="4">
    	0xC03F 0x807F
    </button>
    <button label="" labelSize="12.0" span="4" fillColor="FF000000"></button>
    <button label="&#1045593;" labelSize="25.0" span="4">
    	0xC03F 0xB04F
    </button>
    <button label="&#1045340;" labelSize="25.0" span="4">
    	0xC03F 0x7887
    </button>
    <button label="&#1044998;" labelSize="25.0" span="4">
    	0xC03F 0xF807
    </button>
    <button label="&#1045364;" labelSize="25.0" span="6">
    	0xC03F 0xA857
    </button>
    <button label="&#1045525;" labelSize="25.0" span="6">
    	0xC03F 0x8877
    </button>
  </device>
</irplus>
```

I want to convert it into ir file, but think my hex don't work, is the convert source of error ?
```ir
Filetype: IR signals file
Version: 1
# 
# Artlii YG300 Projector
#
name: Power
type: parsed
protocol: NEC
address: C0 3F 00 00
command: B8 47 00 00
#
name: Mute
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 40 BF 00 00
#
name: Up
type: parsed
protocol: NEC
address: C0 3F 00 00
command: E0 1F 00 00
# 
name: Down
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 98 67 00 00
# 
name: Right
type: parsed
protocol: NEC
address: C0 3F 00 00
command: A0 5F 00 00
# 
name: Left
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 90 6F 00 00
# 
name: Enter
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 85 7A 00 00
# 
name: Menu
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 9A 65 00 00
# 
name: Vol+
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 98 67 00 00
# 
name: Vol-
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 99 66 00 00
# 
name: Home
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 95 6A 00 00
#
name: src/HDMI
type: parsed
protocol: NEC
address: C0 3F 00 00
command: 70 8F 00 00
```

Thanks.

Sigri44, 2023-02-18:
Yeeaaaahhh very good job ! Thx

But my first code example don't in official irplus but in extension. Can you ping dev to add this code into new file please ?

UberGuidoZ, 2023-02-22:
You can find the code used for the conversion through the link. =)

https://github.com/sasiplavnik/irplus2flipper

Sigri44, 2023-03-01:
> You can find the code used for the conversion through the link. =)
> 
> https://github.com/sasiplavnik/irplus2flipper

Thx !

After test, i think don't work correctly !?

[image]

Sigri44, 2023-03-07:
Fixed in : https://github.com/Lucaslhm/Flipper-IRDB/pull/435

closed 2023-03-07 by Sigri44