Lucaslhm/Flipper-IRDB #494: Xaiomi Mi TV Stick

MrFarePlay, 2023-05-20:
Can you please add the remote file for the Xaiomi Mi TV Stick?? The MiBox file doesn't work with the Mi TV Stick, and for some reason, my flipper can't read the IR signal from my Mi Stick remote.

LupusE, 2023-06-04:
All Xaiomi Mi TV remotes I've had hands on working over Bluetooth. Maybe there are other models with IR as well, but I would not bet on that.

MrFarePlay, 2023-06-04:
> All Xaiomi Mi TV remotes I've had hands on working over Bluetooth. Maybe there are other models with IR as well, but I would not bet on that.

That's correct. My mistake not sure how I missed it. This one does work over Bluetooth.

Still, even with flipper the Mi stick couldn't detect it as a Bluetooth device, even while using the flipper Bluetooth controllers.

closed 2023-06-04 by MrFarePlay