Lucaslhm/Flipper-IRDB #597: Request: Generic LG TV codes from a PDF

stek29, 2023-10-15:
this PDF has a lot of Generic LG TV codes, including one I needed and haven't found in this repo -- one to start service mode.
http://www.proaudioinc.com/Dealer_Area/RS232.pdf

see ALL GENERIC LG IR HEX CODES FOR TELEVISION AND MONITORS
complete hex code 04FB 08F7 -> address: 04 command: 08

nortakales, 2024-01-29:
I don't see "service mode" in that PDF when searching, but do see "service menu". Looks like the command for that is FB though, not 08. 08 is the Power command. I am currently working on files for two separate LG remotes I have, and will also have files for any other commands I can find (which this PDF helps with) that aren't on the remotes. For now though, give this a try:

```
#
name: Service Menu
type: parsed
protocol: NEC
address: 04 00 00 00
command: FB 00 00 00
```

stek29, 2024-03-07:
@nortakales I have just filled this issue so it's not lost and LG remote is created in this repo, I have got it working long time ago manually using this file

```
Filetype: IR signals file
Version: 1
#
name: ServiceMenu_Start
type: parsed
protocol: NEC
address: 04 00 00 00
command: FB 00 00 00
#
name: ServiceMenu_Adj
type: parsed
protocol: NEC
address: 04 00 00 00
command: FF 00 00 00
#
name: ServiceMenu_Stop
type: parsed
protocol: NEC
address: 04 00 00 00
command: FA 00 00 00
```

so you're correct, it's FB and not 08