Lucaslhm/Flipper-IRDB #632: How can I have the power_off all on

samuellazea, 2023-12-19:
Hello,

How can I have the power_off all ir file working? I tried to follow the instructions from 
https://github.com/Lucaslhm/Flipper-IRDB/tree/main/_Converted_#to-install-power_off-remote-on-your-flipper   but I get the following error.

[image]

Any help would be greatly appreciated

UberGuidoZ, 2023-12-20:
It's very possible the universal remote has been updated since this file was made and is looking for something else. But it's also possible the file you downloaded did not download correctly. Try saving the file [from this link](https://raw.githubusercontent.com/Lucaslhm/Flipper-IRDB/main/_Converted_/Power_off.ir) and see if it works better.

samuellazea, 2023-12-20:
It has the same issue. 

[image]

Is there any way I could get a debug log for this so I can help with investigation? I do have debug enabled on the device but I don't see anything relevant to running of the script

`64430261 [RPC] (7) Storage List @/ext/infrared START
64430354 [RPC] (7) Storage List @/ext/infrared SUCCESS
64451431 [RPC] (8) Storage Remove @/ext/infrared/Power_off_ALL.ir START
64451495 [RPC] (8) Storage Remove @/ext/infrared/Power_off_ALL.ir SUCCESS
64451495 [RPC] (9) Storage List @/ext/infrared START
64451583 [RPC] (9) Storage List @/ext/infrared SUCCESS
64467289 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
64467635 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
64488636 [RPC] (10) Gui Start ScreenStream START
64488689 [RPC] (10) Gui Start ScreenStream SUCCESS
64496918 [RPC] (11) Gui Send Input START
64496975 [RPC] (11) Gui Send Input SUCCESS
64496976 [RPC] (12) Gui Send Input START
64497034 [RPC] (12) Gui Send Input SUCCESS
64497034 [RPC] (13) Gui Send Input START
64497091 [RPC] (13) Gui Send Input SUCCESS
64497091 [RPC] (14) Gui Send Input START
64497151 [RPC] (14) Gui Send Input SUCCESS
64497151 [RPC] (15) Gui Send Input START
64497211 [RPC] (15) Gui Send Input SUCCESS
64497212 [RPC] (16) Gui Send Input START
64497270 [RPC] (16) Gui Send Input SUCCESS
64509328 [RPC] (17) Gui Send Input START
64509381 [RPC] (17) Gui Send Input SUCCESS
64509422 [RPC] (18) Gui Send Input START
64509477 [RPC] (18) Gui Send Input SUCCESS
64509478 [RPC] (19) Gui Send Input START
64509534 [RPC] (19) Gui Send Input SUCCESS
64534464 [qt.qpa.mime] Retrying to obtain clipboard.`

samuellazea, 2024-01-02:
@UberGuidoZ  do you have any idea on this please? :)

UberGuidoZ, 2024-01-22:
When you save the file I linked, you need to follow the same directions about renaming it and replacing the projectors IR file. From your screenshot, it appears that was not done this time.

kdpdev, 2024-02-25:
> It's very possible the universal remote has been updated since this file was made and is looking for something else. But it's also possible the file you downloaded did not download correctly. Try [...]

There is strange 1 strange signal:
```
name: POWER
type: parsed
protocol: NECext
address: 04 01 00 00
command: 344 00 00 00

```

command: **344** 00 00 00

UberGuidoZ, 2024-04-01:
@kdpdev Where are you seeing the incorrect signal?

kdpdev, 2024-04-17:
@UberGuidoZ , go to the link that had provided by you: [https://raw.githubusercontent.com/Lucaslhm/Flipper-IRDB/main/_Converted_/Power_off.ir](url)
Try to find there the string: command: 344 00 00 00
344 is not wellformed byte of hex format. There are should be exactly 2 symbols in range 0-9 or a-f or A-F