Lucaslhm/Flipper-IRDB #735: Merge Categories: Audio_Receivers and Audio_Video_Receivers

MDcurIOs1ty, 2024-04-07:
I believe the idea here was to separate **_A/V boxes_** (e.g., those used to power surround sound systems as well as connect to TVs, DVD/Blu-ray players, game consoles, etc.) and **_stereo receivers_** (e.g., those used to power passive bookshelf speakers). The problem is that both folders seem to consist of mostly A/V receivers (after randomly choosing a few and looking up the models). I agree with the separation, but to save you the headache of having to keep track of whether the remotes are for A/V or audio-only receivers (unless you or someone else wants to start), it may be easier to merge the two folders and call it something like Audio_and_Video_Receivers (or whatever else you think is appropriate).

UberGuidoZ, 2024-05-03:
Excellent suggestion, will likely do so and push it up from my repo since it's a bit easier for me to process it that way. Thank you!

UberGuidoZ, 2024-05-03:
Done!

closed 2024-05-03 by UberGuidoZ