Lucaslhm/Flipper-IRDB #771: Missing very common AC brand. (Maytag)

iluvflipperzero, 2024-07-10:
I own a Maytag model m6x06f2a AC unit in my room and I got the AC without a remote, I tried searching everywhere for the Maytag ac IR codes in HEX or whatever form but can't find anything. Can someone please upload the file, that would be super nice :) "ive been standing up to turn on this AC for two years and just noticed it had an IR receiver".

jaylikesbunda, 2024-09-19:
It's your lucky day!
Your AC was in the Harmony Remote db :)

(just remember to take the .txt off the file)
[MayTag_M6X06F2A.ir.txt](https://github.com/user-attachments/files/17057724/MayTag_M6X06F2A.ir.txt)

UberGuidoZ, 2024-09-24:
Request merged in https://github.com/Lucaslhm/Flipper-IRDB/pull/799

closed 2024-09-24 by UberGuidoZ