Lucaslhm/Flipper-IRDB #772: [Q] convert codes parsed with tasmota rereceiv

eadmaster, 2024-07-11:
how can i convert codes [received via Tasmota](https://tasmota.github.io/docs/Tasmota-IR/#receiving-ir-commands)?

They follow this format:
`{"IrReceived":{"Protocol":"<value>","Bits":<value>,"Data":<value>}}`

e.g.:
`{"Protocol":"NEC","Bits":32,"Data":"0x20DF10EF"}`

eadmaster, 2024-07-13:
just found out there is [this method](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#ab2e1ce918e4e06b955c3d2a089ce189c)  to perform the conversion.

UberGuidoZ, 2024-08-09:
Certainly hope to see some additions from you! =) Thank you for coming back with the answer.

closed 2024-08-09 by UberGuidoZ

eadmaster, 2024-08-09:
For pc-side use i've made this script
https://gist.github.com/eadmaster/5c36fd467cf57f5f4025ee6b8082a600