Lucaslhm/Flipper-IRDB #776: Add Canon RC-6 universal remote

frank42hh, 2024-07-24:
Hi,

not sure if one of the already present Canon camera remotes does contain this signal, but just in case, here's the IR file for the Canon Universal Remote RC-6.

compatible products:

EOS 5DS / 5DS REOS 600DEOS 7DEOS 450DEOS 550DEOS 5D Mark IIIEOS 90DEOS D30EOS M5EOS 10DEOS 400DEOS 500DEOS 6DEOS 700DEOS 77DEOS 80DEOS 20DEOS 750D /760DEOS D60EOS M3EOS 100DEOS 300DEOS 5D Mark IVEOS 650DEOS 800DEOS M/ EOS M2EOS 350DEOS 60DEOS 60DaEOS 6D Mark IIEOS 7D IIEOS 5D Mark IIEOS 70DEOS M6EOS M5

[Canon RC-6.ir.zip](https://github.com/user-attachments/files/16360282/Canon.RC-6.ir.zip)

UberGuidoZ, 2024-08-09:
Thank you! I'll get it added and labeled.

UberGuidoZ, 2024-08-09:
Done!

closed 2024-08-09 by UberGuidoZ