Lucaslhm/Flipper-IRDB #782: case-sensitive spelling of directories Humidifiers/HoMedics and /HumidifiersHomedics causes problems on case-insensitive file systems

Abdull, 2024-08-24:
There exist directories
* https://github.com/Lucaslhm/Flipper-IRDB/tree/33d7bcead9b858796a34fc3fc261d26b2ad02dca/Humidifiers/HoMedics
* https://github.com/Lucaslhm/Flipper-IRDB/tree/33d7bcead9b858796a34fc3fc261d26b2ad02dca/Humidifiers/Homedics

This causes problems on case-insensitive filesystems such as NTFS.

E.g. I noticed my Linux-Windows syncthing setup failing to synchronize because of this.

Solution would be to use only one capitalization convention.

UberGuidoZ, 2024-09-04:
Makes sense and agreed - any differences like that should absolutely be merged/changed. If you want to take a crack at it, more power to you. =) Otherwise I'll see if I can run through and find them all "Soon ™️ "

UberGuidoZ, 2024-09-27:
Fixed the example above, will look for more as time allows.