Lucaslhm/Flipper-IRDB #810: Panasonic th-50cq1u

BeHaCh, 2024-10-10:
I tried all the listed remotes for Panasonic but none work for this model. I might be able to get access to the remote and do it myself

BeHaCh, 2024-10-10:
Here's the general Panasonic remote. I've tested it and it works for several TH- models.
[Panasonic.txt](https://github.com/user-attachments/files/17333738/Panasonic.txt)

jaylikesbunda, 2024-10-13:
> Here's the general Panasonic remote. I've tested it and it works for several TH- models. [Panasonic.txt](https://github.com/user-attachments/files/17333738/Panasonic.txt)

is this the file for your TV (Panasonic th-50cq1u)

BeHaCh, 2024-10-13:
It's the file I got by copying a generic Panasonic remote, and works on several different th-... model tvs.

jaylikesbunda, 2024-10-14:
> It's the file I got by copying a generic Panasonic remote, and works on several different th-... model tvs.

so it does or doesn't work for Panasonic th-50cq1u?