Lucaslhm/Flipper-IRDB #841: Adding new device: LG OLED55B9PTA TV

hhienuwu, 2024-11-24:
Hi, I would like to ask for adding a new device infrared code to the repository.
Device type: TV
Brand: LG
Model name: LG OLED55B9PTA

GitHub doesn't allow me to upload .ir files, so please remove the .txt file name extension.

Sorry for my bad English, and have a nice day :)

[LG_OLED55B9PTA.ir.txt](https://github.com/user-attachments/files/17893505/LG_OLED55B9PTA.ir.txt)

darmiel, 2024-11-25:
Thank you! It seems like it has the same remote as the `LG_OLED65C8PUA`, so I've added a comment to that file and added the `Movies` button to the remote.

hhienuwu, 2024-11-25:
> Thank you! It seems like it has the same remote as the `LG_OLED65C8PUA`, so I've added a comment to that file and added the `Movies` button to the remote.

Thanks! After some research, it looks like the `LG_OLED65C8PUA` remote looks a bit different from the `LG_OLED55B9PTA`, but they have the same IR code in common.

PR #842 (2024-11-25): Update LG_OLED65C8PUA.ir
closes #841