Lucaslhm/Flipper-IRDB #862: Trying to create .IR file for Wall Clock

mnopneal, 2025-01-03:
We have a big West Ocean digital wall clock which hangs up high on the wall in our auditorium.  And of course we have lost the remote.  I contacted the manufacturer, and got their list of codes.   Here is a bit of it: 

We use NEC protocol, and Address is 00.  
#define RC_ONOFF 0xA25D
#define RC_MUTE  0xE21D
#define RC_BACK  0x629D
#define RC_PLAY  0x9867

So, I thought I'd whip up a .IR file, and load it onto the SD: 
Filetype: IR signals file
Version: 1
# 
name: OnOff
type: parsed
protocol: NEC
address: 00 00 00 00
command: A2 5D 00 00
#
name: Mute 
type: parsed
protocol: NEC
address: 00 00 00 00
command: E2 1D 00 00
#
name: Back 
type: parsed
protocol: NEC
address: 00 00 00 00
command: 62 9D 00 00
#
name: Play 
type: parsed
protocol: NEC
address: 00 00 00 00
command: 98 67 00 00
#

When I pick that remote, the buttons all load.   When I press a button, it says "Failed to Load OnOff".   I'm drawing a blank on what it is objecting to.  

Thoughts?

mnopneal, 2025-01-03:
Dang. Maybe I figured this out.   Basic NEC protocol only allows for commands 0 through 255?   And the values they gave me suggest they really meant to say NECext?   That appears to load and transmit correctly.   I'll have to go and check it out with the actual clock.    Then I can submit this file.

bengtmartensson, 2025-01-03:
Try

```name: OnOff
type: parsed
protocol: NEC
address: 00 00 00 00
command: A2 00 00 00
```

as well as

```name: OnOff
type: parsed
protocol: NEC
address: 00 00 00 00
command: 45 00 00 00
```

mnopneal, 2025-01-03:
I think I will also contact the manufacturer again to seek clarification on NEC versus NECext

closed 2026-03-22 by UberGuidoZ