Lucaslhm/Flipper-IRDB #900: Ch_next and Ch_prev are no longer default

Pandapip1, 2025-02-16:
It seems that the new names that are used by default are `Ch_up` and `Ch_down`. You should probably add a lint to enforce the new or old behaviour.

Pandapip1, 2025-02-16:
Interesting. My stock remote app uses Ch_up and Ch_down, but XRemote uses the old values. I assume you want the XRemote style.

Pandapip1, 2025-02-16:
Looks like there's also a discrepancy with fast forward/back:

* `Fast_fo` vs `FF`
* `Fast_ba` vs `Rew`

I see a lot of instances of both, and in the second instance, the full word `Rewind`.

Pandapip1, 2025-02-16:
I also see "Enter" used many times when "Ok" should be used instead

Pandapip1, 2025-02-16:
I also see `VOL+` and `VOL-` used instead of `Vol_up` and `Vol_dn`, `MUTE` used instead of `Mute`, `CH-` and `CH+` used instead of `Ch_prev` and `Ch_next`, and `POWER` used instead of `Power`.

bengtmartensson, 2025-02-16:
Personally, I would like to ban "power" (including all capitalization variants), and use power_toggle, power_on, power_off. The same goes for "mute". 

Also note that many devices have no separate play and pause commands (should be pause_toggle really), but a combined play/pause.

Pandapip1, 2025-02-17:
> Also note that many devices have no separate play and pause commands (should be pause_toggle really), but a combined play/pause.

The standard for that seems to be `Play_pa`