Lucaslhm/Flipper-IRDB #927: complete captures of mini Hi-Fi audio systems IR files

vlado2222, 2025-04-30:
Hi.

I made complete captures of remotes for mini Hi-Fi audio systems Panasonic SC-AK15 (Year 1998) (remote: RAK-CH940WK) and Philips FW-M55 (Year 2000) (Remote: Philips RC2519/01).

Note for Philips FW-M55: play/pause/next/prev/fast_forward/fast_rewind are different codes for CD/TAPE/Tuner so they have to be three times and named CD_Next, Tuner_next etc.

[Panasonic_SC-AK15.zip](https://github.com/user-attachments/files/19969070/Panasonic_SC-AK15.zip)
[Philips_FW-M55.zip](https://github.com/user-attachments/files/19969069/Philips_FW-M55.zip)

I suggest to create new directory for mini Hi-Fi audio systems.

Enjoy!

vlado2222, 2025-04-30:
One more remote to add: Sony RM-SC30 used for many different Sony mini Hi-Fi systems.

[Sony_CMT-EH10.zip](https://github.com/user-attachments/files/19970750/Sony_CMT-EH10.zip)

UberGuidoZ, 2025-08-09:
Hi! We’ve just re-licensed the project under CC0-1.0 to make the IR-code data freely reusable for everyone. (https://github.com/Lucaslhm/Flipper-IRDB/pull/960, https://github.com/Lucaslhm/Flipper-IRDB/pull/843)
Because your PR contains work created before the switch, we need your explicit OK before merging.

Please add a comment (or a commit with a Signed-off-by: line) saying:

`I confirm that my contributions in this PR are released under CC0-1.0.`

If you have any questions, let us know. Thanks!

vlado2222, 2025-08-09:
I confirm that my contributions in this PR are released under CC0-1.0.

UberGuidoZ, 2026-03-22:
Done! Thank you.