Lucaslhm/Flipper-IRDB #929: New IR file for the MAG TV boxes

vlqdus, 2025-05-24:
i used my esp32 to record and save the new IR RAW code of some necessary buttons, i would like to upload them to this repo.

UberGuidoZ, 2025-08-09:
I confirmed the linked files are identical

UberGuidoZ, 2025-08-09:
Hi! We’ve just re-licensed the project under CC0-1.0 to make the IR-code data freely reusable for everyone. (https://github.com/Lucaslhm/Flipper-IRDB/pull/960, https://github.com/Lucaslhm/Flipper-IRDB/pull/843)
Because your PR contains work created before the switch, we need your explicit OK before merging.

Please add a comment (or a commit with a Signed-off-by: line) saying:

`I confirm that my contributions in this PR are released under CC0-1.0.`

If you have any questions, let us know. Thanks!